worproject/rpi5-uefi #1: Device tree support

leo60228, 2024-01-04:
I understand that this is a fairly niche usecase, but it would be helpful to have support for a device tree-based boot process when using a downstream kernel (with the patches from raspberrypi/linux). This would allow Linux distributions to have complete hardware support in an installed system while still using a PC-like boot and installation process: users would just download and use an installer image with a mainline kernel like normal, and then could install a linux-rpi kernel package to improve hardware support.

leo60228, 2024-01-05:
I tried enabling FdtDxe and removing the PcdSystemTableMode check, but this results in a black screen. I assume there's debug output, but I don't currently have a cable for the debug UART.

mariobalanica, 2024-01-05:
Available in the latest version.

closed 2024-01-05 by mariobalanica