worproject/rpi5-uefi #4: Keyboard Not detected with Original Adaptor supply

amyreza, 2024-01-28:
My keyboard is not detected when I want to press the ESC key and I am confused about what to do. I've changed 4 brands of keyboard and moved from USB 1 - 4 on Raspi 5 but it still doesn't work when I want to install Proxmox . I use the official power supply and the unit I just bought

FriendlyNGeeks, 2024-01-28:
Use right arrow on keyboard from first menu and dont ask me why it works,
lolz

amyreza, 2024-01-28:
I've tried, it still doesn't work bro.
Is there another way?

Please help

FriendlyNGeeks, 2024-01-28:
first make sure you following this guide and grabbing the right image https://youtu.be/oe1_JVl63a0?si=d4FP9gVdbyKVQKGM . Next I would ask that you provide picture of which screen you are stuck on.  If you are trying to get into UEFI boot order might need to be updated first using Raspbian and running sudo raspi-config > find boot and make sure sd is first or usb if your booting from usb. Also I stand corrected about side arrow that was the work around for the full pc proxmox.

amyreza, 2024-01-28:
Yes. I also followed that guide. Initially, by simply extracting the RPI5 UEFI file. My keyboard cannot be detected to enter the RPI5 Boot menu

And now I have succeeded in a way
Flash the SD CARD with the Ubuntu server using Raspi Imager

and replace the file on the SDCARD with this RPI5 UEFI extract and VIOLA. New keyboard works and already successfully installed proxmox .

Now a new problem arises, namely after installing Proxmox. Proxmox OS can't detect LAN and wifi.
I have tried "sudo dhclient -v"

still can't find the LAN and wifi drivers.

Can you help me?
I'll make a new ticket, guys

wb7odyfred, 2024-02-16:
> Now a new problem arises, namely after installing Proxmox. Proxmox OS can't detect LAN and wifi.
> I have tried "sudo dhclient -v"

```
sudo ifconfig genet0
sidp dhclient genet0
sudo ifconfig genet0

```
Other commands that maybe useful.  
```
sudo service netif restart
sudo service routing restart
ifconfig genet0
netstat -nr
sockstat -4

```
ifconfig wlan0
sudo dhclient wlan0
```

[https://ghostbsd-arm64.blogspot.com/2024/01/january-18-2024-more-random-notes-on.html](https://ghostbsd-arm64.blogspot.com/2024/01/january-18-2024-more-random-notes-on.html)
Add a 4 lines to  /boot/loader.conf file
# WIFI load   
rtwn_load="YES"  
if_rtwn_usb_load="YES"  
legal_realtek_license_ack=1 

Add 2 lines to  /etc/rc.conf   file
wlands_rtwn0="wlan0"
 ifconfig_rtwn0="DHCP"  # for open wifi access point or  "WPA DHCP" for encrypted wifi access point connection

```

```
ifconfig wlan0 create rtwn0 wlandev wlan0
ifconfig wlan0 ssid FreshTomato24
dhclient wlan0

```
#file /etc/wpa_supplicant.conf 

# Open network wifi access point.   I recommend FreshTomato.org for router software
edit file  /etc/wpa_supplicant.conf
network={
          ssid="FreshTomato24"
          key_mgmt=NONE
}
# WPA-DHCP example
network={
         ssid="simple"
         psk="very secret passphrase"
         proto=WPA RSN
         priority=5
}

# https://www.ccammack.com/posts/configure-a-wifi-client-on-freebsd/

closed 2024-03-15 by mariobalanica