worproject/rpi5-uefi #5: AFTER INSTALLING PROMOX with RPI5 UEFI . LAN and WIFI RPI5 cannot detected

amyreza, 2024-01-28:
Yes. I also followed that guide. Initially, by simply extracting the RPI5 UEFI file. My keyboard cannot be detected to enter the RPI5 Boot menu

And now I have succeeded in a way
Flash the SD CARD with the Ubuntu server using Raspi Imager

and replace the file on the SDCARD with this RPI5 UEFI extract and VIOLA. New keyboard works and already successfully installed proxmox .

Now a new problem arises, namely after installing Proxmox. Proxmox OS can't detect LAN and wifi.
I have tried "sudo dhclient -v"

still can't find the LAN and wifi drivers.

Can you help me?
I'll make a new ticket, guys

FriendlyNGeeks, 2024-01-29:
Video tells you the onboard wifi and NIC drivers have not been enabled in
this UEFI as of yet. You will need to either use a USB to ethernet or...
connect your phone via USB and enable USB tethering on the phone

amyreza, 2024-01-29:
OK. I hope in the future. LAN and onboard wifi work well for updates. Ok, I will try to buy LAN to USB this afternoon. Thank you guys

amyreza, 2024-01-29:
I bought a USB 2.0 LAN from the M-TECH brand but Proxmox still can't detect my LAN driver. even though the USB LAN light is on. I set the IP in Proxmox but it still doesn't read it

mariobalanica, 2024-01-31:
Wi-FI not working is likely due to the driver not finding the correct BCM firmware. This is easy to fix, you can find details online. 
`dmesg` should show exactly why it doesn't load.

As for Ethernet, that requires both firmware and OS support. I personally don't have any plans to submit the necessary Linux patches for it (but contributions are welcome!).

However, you can get most hardware working by using an appropriate kernel and device tree pair, for instance the downstream RPi kernel with DTB included here.

closed 2024-02-09 by mariobalanica