worproject/rpi5-uefi #7: Rpi 5 with Red Hat Enterprise Linux 9.3. Ethernet and Wifi devices not working.

mugicoder13, 2024-02-09:
I installed RHEL 9.3 on Raspberry Pi 5 by following the  instructions on setting up the UEFI. I successfully installed the RHEL 9.3 server (Not GUI). During installation with the graphical setup wizard I used USB network adapter for internet connection. After the installation when logged in I saw that the USB network adapter was down. 
I set the Device Manager->Raspberry Pi Configuration->ACPI / Device Tree as Both during installation.

What possibly could be the reason for not able to use the USB network adapter, which I used the same during the installation ?

Since I can't able to find solution, I connected via mobile USB tethering. Now I can able to use the internet, but I need the onboard Ethernet and Wifi.

mugicoder13, 2024-02-09:
RHEL 9.3 in Rpi 5 - Network driver issue

mariobalanica, 2024-02-09:
See #5

closed 2024-02-09 by mariobalanica