worproject/rpi5-uefi #9: Unusual behavior across arm cortex cpus

keixthb, 2024-02-17:
## Steps to reproduce:

Install red hat enterprise linux (RHEL) 9.0 on a raspberry pi4b, following my steps outlined [here](https://github.com/keixthb/rhel_pi4).

### Using pi4b
Installation goes perfectly using ```RPi4_UEFI_Firmware_v1.29```and we can log into the desktop and update kernel (visit my repo to see results). Using ```RPi4_UEFI_Firmware_v1.35``` installation succeeded, but boot FAILS with a synchronous exception. 

### Using pi5b
I install on a raspberry pi 5 using the exact same methods as for the pi4b, except using [this](https://github.com/worproject/rpi5-uefi/releases) UEFI, and the boot fails in a very similar way as the raspberry pi 4b did with the latest 1.35 [firmware](https://github.com/pftf/RPi4/releases).

## Error Message:

synchronous exception 0x376302EC

## Information on your system:

Tested with both 64gb and 256gb sd cards--same result. The onboard Ethernet for the pi5 did not work with the RHEL installer, so I had to use a usb-ethernet dongle to log into red hat. The firmware is always able to boot the .iso installer, and the installation goes great so I have no idea what could be going wrong. Also, it works perfectly with the pi4b, so I wonder if there were changes introduced after 1.29 that are breaking  things? Can anyone shed some light on this issue? Or can anyone replicate this error?

My goal is to get RHEL 9.3 running on the pi5b. Thanks in advance.

mariobalanica, 2024-02-17:
https://github.com/worproject/rpi5-uefi?tab=readme-ov-file#linux

keixthb, 2024-02-17:
This worked great! Thank you so much! Is there anything I can do to enable the pcie functionality? ```lspci``` returns nothing from the console, even though I have a device connected. 

[image]

mariobalanica, 2024-02-17:
Nice!

PCIe is not supported yet, but I ordered an adapter and will look into it soon.

closed 2024-02-17 by mariobalanica

keixthb, 2024-02-18:
@mariobalanica Cool! I'll write a new tutorial for rhel 9.3 on the pi5b in the next few days and update with a link here.

keixthb, 2024-02-19:
@mariobalanica after resizing the bar (```pci=realloc=on``` in grub), the pi went into emergency mode with dracut. I solved this using the UEFI, and going to:

```Raspberry Pi Configuration```->```ACPI / Device Tree```->```System Table Mode <ACPI>``` && ```Compatibility Mode <Full Bay Trail>```

The original synchronous error went away, and I was able to boot back into rhel no issues. 

**Note** The onboard ethernet still does not work (ethernet with the dongle still does), and ```lspci``` still returns nothing. 

Here is my ```dmesg```:

```console
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 5.14.0-362.18.1.el9_3.aarch64 (mockbuild@arm64-026.build.eng.bos.redhat.com) (gcc (GCC) 11.4.1 20230605 (Red Hat 11.4.1-2), GNU ld version 2.35.2-42.el9) #1 SMP PREEMPT_DYNAMIC Wed Jan 3 18:51:55 EST 2024
[    0.000000] The list of certified hardware and cloud instances for Red Hat Enterprise Linux 9 can be viewed at the Red Hat Ecosystem Catalog, https://catalog.redhat.com.
[    0.000000] efi: EFI v2.70 by worproject
[    0.000000] efi: ACPI 2.0=0x37e70018 SMBIOS=0x37d00000 SMBIOS 3.0=0x37ce0000 MEMATTR=0x39f64018 RNG=0x3b3dd298 MEMRESERVE=0x37f43b98 
[    0.000000] efi: seeding entropy pool
[    0.000000] efi: memattr: Unexpected EFI Memory Attributes table version 2
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x0000000037E70018 000024 (v02 RPIFDN)
[    0.000000] ACPI: XSDT 0x0000000037E7FE98 000054 (v01 RPIFDN RPI5     00000200      01000013)
[    0.000000] ACPI: FACP 0x0000000037E7FB98 000114 (v06 RPIFDN RPI5     00000200 EDK2 00000300)
[    0.000000] ACPI: DSDT 0x0000000037E7EC18 000875 (v02 RPIFDN RPI5     00000002 INTL 20200925)
[    0.000000] ACPI: DBG2 0x0000000037E7FA98 000061 (v00 RPIFDN RPI5     00000200 EDK2 00000300)
[    0.000000] ACPI: GTDT 0x0000000037E7FD18 000068 (v03 RPIFDN RPI5     00000200 EDK2 00000300)
[    0.000000] ACPI: APIC 0x0000000037E7E998 000184 (v05 RPIFDN RPI5     00000200 EDK2 00000300)
[    0.000000] ACPI: PPTT 0x0000000037E7F698 000130 (v02 RPIFDN RPI5     00000200 EDK2 00000300)
[    0.000000] ACPI: BGRT 0x0000000037E7FF98 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.000000] NUMA: Failed to initialise from firmware
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x1feff8ec0-0x1feffdfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000]   Device   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000020ffff]
[    0.000000]   node   0: [mem 0x0000000000210000-0x0000000037c2ffff]
[    0.000000]   node   0: [mem 0x0000000037c30000-0x0000000037c7ffff]
[    0.000000]   node   0: [mem 0x0000000037c80000-0x0000000037c80fff]
[    0.000000]   node   0: [mem 0x0000000037c81000-0x0000000037e6ffff]
[    0.000000]   node   0: [mem 0x0000000037e70000-0x0000000037e7ffff]
[    0.000000]   node   0: [mem 0x0000000037e80000-0x0000000037f3ffff]
[    0.000000]   node   0: [mem 0x0000000037f40000-0x000000003b21ffff]
[    0.000000]   node   0: [mem 0x000000003b220000-0x000000003b2affff]
[    0.000000]   node   0: [mem 0x000000003b2b0000-0x000000003b2bffff]
[    0.000000]   node   0: [mem 0x000000003b2c0000-0x000000003b3dffff]
[    0.000000]   node   0: [mem 0x000000003b3e0000-0x000000003f3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 3072 pages in unavailable ranges
[    0.000000] crashkernel reserved: 0x00000000ec000000 - 0x0000000100000000 (320 MB)
[    0.000000] psci: probing for conduit method from ACPI.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.4
[    0.000000] percpu: Embedded 32 pages/cpu s92840 r8192 d30040 u131072
[    0.000000] pcpu-alloc: s92840 r8192 d30040 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[... 465 lines cut ...]
[    6.670123] Bluetooth: SCO socket layer initialized
[    6.672568] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2
[    7.683562] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[    7.871072] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Quota mode: none.
[    8.839219] RPC: Registered named UNIX socket transport module.
[    8.839224] RPC: Registered udp transport module.
[    8.839226] RPC: Registered tcp transport module.
[    8.839228] RPC: Registered tcp NFSv4.1 backchannel transport module.
[... cut ...]
[   17.743541] systemd-journald[474]: Oldest entry in /run/log/journal/69b5b0541be94047967a258b5aade1ee/system.journal is older than the configured file retention duration (1month), suggesting rotation.
[   17.743547] systemd-journald[474]: /run/log/journal/69b5b0541be94047967a258b5aade1ee/system.journal: Journal header limits reached or header out-of-date, rotating.
[   28.276817] rfkill: input handler disabled
[   36.885131] rfkill: input handler enabled
[   40.122933] rfkill: input handler disabled

```

keixthb, 2024-02-19:
@mariobalanica Did a reboot and something interesting happened:

This is the output of ```dmesg | grep -i pci```

```console
[    0.007015] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    0.039041] PCI: CLS 0 bytes, default 64
```