worproject/rpi5-uefi #10: Unusual behavior using Red Hat Enterprise Linux (RHEL) 9.3 on a pi5b

keixthb, 2024-02-19:
## Steps to reproduce:

Install red hat enterprise linux (RHEL) 9.3 on a raspberry pi5b, following my steps outlined [here](https://github.com/keixthb/rhel_pi4). Make sure to change the UEFI settings accordingly. 

Resize the base address register (bar) by editing ```/etc/default/grub``` to include:

```console
pci=realloc=on
```

update grub with:

```console
sudo grub2-mkconfig -o /boot/efi/EFI/redhat/grub.cfg
```

...and reboot. Then, use ```lspci -M``` (case sensitive)

## Error Message:

<details><summary>Error message</summary>

pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:01.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:02.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:03.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:04.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:05.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:06.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:07.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:08.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:09.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:0a.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:0b.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:0c.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:0d.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:0e.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:0f.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:10.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:11.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:12.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:13.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:14.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:15.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:16.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:17.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:18.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:19.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:1a.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:1b.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:1c.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:1d.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:1e.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:00:1f.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:00.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:01.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:02.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:03.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:04.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:05.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:06.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:07.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:08.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:09.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:0a.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:0b.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:0c.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:0d.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:0e.0/config
pcilib: Cannot open /sys/bus/pci/devices/0000:01:0f.0/config
[... 354 lines cut ...]
Vulnerability Gather data sampling: Not affected
Vulnerability Itlb multihit:        Not affected
Vulnerability L1tf:                 Not affected
Vulnerability Mds:                  Not affected
Vulnerability Meltdown:             Not affected
Vulnerability Mmio stale data:      Not affected
Vulnerability Retbleed:             Not affected
Vulnerability Spec rstack overflow: Not affected
Vulnerability Spec store bypass:    Mitigation; Speculative Store Bypass disabled via prctl
Vulnerability Spectre v1:           Mitigation; __user pointer sanitization
Vulnerability Spectre v2:           Mitigation; CSV2, BHB
Vulnerability Srbds:                Not affected
Vulnerability Tsx async abort:      Not affected
```

... and my ```spack debug report```:

```console
* **Spack:** 0.22.0.dev0 (d670a2a5ce564fb5386fa55deb987803d12e8de0)
* **Python:** 3.9.18
* **Platform:** linux-rhel9-cortex_a72
* **Concretizer:** clingo
```

My goal is to get my pcie usb controller working in RHEL 9.3 with the pi5b. Thanks in advance.

mariobalanica, 2024-03-15:
PCIe support is now available in the latest build artifact: https://github.com/worproject/rpi5-uefi/actions

Note that in ACPI mode, Linux can only see a single-function PCIe device. Anything more than that cannot be supported without OS changes due to a hardware compliance issue.

keixthb, 2024-03-15:
@mariobalanica what command line arguments did you use with the ```build.sh```?

keixthb, 2024-03-15:
@mariobalanica I tested with the pre built artifact and when I boot with my pci device attached, all my attached USB devices disconnect. Unfortunately, this means that the USB to ethernet dongle also disconnects and I can no longer ssh into the system. Removing my PCI device solves this issue.

keixthb, 2024-03-15:
Calling lspci from the command line in RHEL returns the same result as before.

mariobalanica, 2024-03-16:
> I tested with the pre built artifact and when I boot with my pci device attached, all my attached USB devices disconnect. Unfortunately, this means that the USB to ethernet dongle also disconnects and I can no longer ssh into the system. Removing my PCI device solves this issue.

@keixthb What PCIe device is this? Can you grab the serial logs from a debug build of UEFI?

> Calling lspci from the command line in RHEL returns the same result as before.

For Linux, you need to go to `Device Manager`->`Raspberry Pi Configuration`->`ACPI / Device Tree` and change `ECAM Compatibility Mode` to `AMAZON GRAVITON`.

It probably won't work if it doesn't even show up in UEFI. Still worth a try though, and dmesg output would be helpful.

> what command line arguments did you use with the build.sh?

It doesn't need any arguments to produce a **release** image. For debug builds, append `--debug 1`.

keixthb, 2024-03-16:
@mariobalanica Yes, I can do a debug build & grab the serial logs. What build environment do you support for this? 

My device is a Nvidia Tesla P100 accelerator. I understand that this is an edge case. 

Nonetheless, my current system looks something like this:

- PCIe adapter for a m.2 ssd: [here](https://www.amazon.com/dp/B0CPPGGDQT?psc=1&ref=ppx_yo2ov_dt_b_product_details)
- m.2 riser [here](https://www.amazon.com/dp/B09N1FYCQZ?psc=1&ref=ppx_yo2ov_dt_b_product_details)
- usb mining riser, similar to [this](https://www.amazon.com/FebSmart-Bitcoin-Litecoin-ETH-VER006C-Cable-GPU-Adapter-Ethereum/dp/B07GJNMLYM/ref=sr_1_3?crid=FOOE3UTU5GMM&dib=eyJ2IjoiMSJ9.9x-03TTZmzIqDZ28vXQVuJoElrKAK-ec4X03iLH6-mIYA65Pp0tf6lDJenifwdcJFbsT2r8zk0VhrpFIkWx2NnxpzlXt0IrjBMuRgtH8ZypeDoBaUnDb-vElneH8jB8_NNV0POvDEFkbIduCqMpP1mfzjVy4ewqHrOAoP_lpeP2hpJB28AaDl2d87q_DoqVMHXblaka2GBnx2BU8TQiFKVYiRq-Wlw0g_f5_mA9YMA8.lWW3jUVWu0tcJf5tULaCkpplbw1DEIQmgxuKRgD_otI&dib_tag=se&keywords=gpu+mining+riser&qid=1710548950&sprefix=gpu+mining+rise%2Caps%2C155&sr=8-3)
- regular atx power supply
- nvidia tesla p100
- raspberry pi 5b

[image]

[image]

The tesla card requires an extremely large base address register, which is why I use the ```pci=realloc=on``` parameter with grub. I've tested the risers, and I know that they work with this card on an x86 system. Installing the driver on x86 was tricky, and it required disabling CSM and enabling Above 4G Decoding. 

I know that this card is compatible with the ARM Cortex series of processors, because I've successfully gotten it to show up using ```lspci``` on an AMD ZCU-102 [FPGA](https://www.xilinx.com/products/boards-and-kits/ek-u1-zcu102-g.html). The only issue we had, was that the OS did not use grub, so we could not resize the base address register to communicate with the driver. We were able to get the driver installed successfully, though. But as you would expect, ```nvidia-smi``` returned "no devices found", because of the 32 bit BAR.

keixthb, 2024-03-16:
Just to be clear, the tesla GPU does not have a video out, and I am only interested in using it as a proof-of-concept for parallel computing research.

mariobalanica, 2024-03-16:
> Yes, I can do a debug build & grab the serial logs. What build environment do you support for this?

Nothing fancy. Just an USB to UART adapter set to 115200 baud, connected to the RPI's "special" debug port.
The debug UEFI image can be found in latest Github artifacts.

> The tesla card requires an extremely large base address register, which is why I use the pci=realloc=on parameter with grub. I've tested the risers, and I know that they work with this card on an [...]

This may be the reason. The Pi 5 can support up to 12 GB of 64-bit prefetchable BAR space, while 32-bit non-prefetchable BARs are limited to 1 GB in UEFI, since it has to cut into usable system RAM.

EDK2 downgrades 64-bit BARs to 32-bit if they're not marked as `Prefetchable`, so it's likely running out of space.

mariobalanica, 2024-03-16:
Can you give this a try and see if it works?
[RPI_EFI_mem64_preserve.zip](https://github.com/worproject/rpi5-uefi/files/14621877/RPI_EFI_mem64_preserve.zip)

keixthb, 2024-03-16:
@mariobalanica Absolutely. I'll run it and report back.

mariobalanica, 2024-03-16:
I've updated the archive above, turns out I patched the wrong driver previously.

keixthb, 2024-03-16:
@mariobalanica You solved the mouse and keyboard issue with the mem64 preserve. These were the settings I used:

[image]
[image]

[image]

With these settings, I am able to login with ssh. Running ```lspci``` also returns the same result, both before installing the tesla card:

[image]

...and after installing the card:

[image]

mariobalanica, 2024-03-16:
What's the output of the `pci` command in UEFI shell?

keixthb, 2024-03-16:
@mariobalanica With the updated archive above, the mouse and keyboard lock up again. Without the GPU installed, this is the output I get in the shell:

[image]

With the GPU installed, I cannot access the shell or the desktop (because none of the USB devices work). I lose ethernet as well, so no ssh either. But on boot, I get this message:

[image]

mariobalanica, 2024-03-16:
Hmm, what about the previous archive? What does `pci` say?

keixthb, 2024-03-16:
It would appear that the results are the same. With the previous archive, this was the result before installing the card:

[image]

This is the result after installing the card:

[image]

I will note that the keyboard and mouse did not work until altering the settings for ```ACPI / Device Tree```, ```PCI Express``` and ```EFI Memory Attribute Protocol``` as defined above.

mariobalanica, 2024-03-16:
> I will note that the keyboard and mouse did not work until altering the settings for ACPI / Device Tree, PCI Express and EFI Memory Attribute Protocol as defined above.

Is the PCIe controller disabled in the settings now?

If the archive above doesn't work then I'm not sure what's causing this and would have to look at UART logs.

keixthb, 2024-03-16:
@mariobalanica I must've had some settings wrong with the updated archive. I tested a second time, and after changing the settings I get the exact same results as with the previous archive. The output of ```pci``` in the shell is also the exact same, with the previous archive and the new one.

mariobalanica, 2024-03-16:
Yeah, not sure what it doesn't like about this PCIe device. Debug logs from the serial port should give an insight.

mariobalanica, 2024-04-03:
Closing this because neither NVIDIA proprietary or nouveau drivers work with RPi's PCIe, no matter if it's ACPI or FDT.

closed 2024-04-03 by mariobalanica