worproject/rpi5-uefi #12: Support of total_mem assignment in config.txt ?

markmi, 2024-03-16:
https://www.raspberrypi.com/documentation/computers/config_txt.html#total_mem documents using an assignment to total_mem in config.txt to test handling RAM sizes for which one may not have devices with that size. I've used that in the past on RPi*'s (but generally in a U-Boot UEFI context).

I tried a total_mem=2048 for a UEFI/ACPI boot but the 8 GiByte RPi5 still booted as having 8 GiBytes. Thus I cannot test how things go for smaller RAM sizes on the RPi5. (Nice, in part, because it takes less time for various kinds of experiments than the older RPi*'s.) Also, I wonder if the RPi* firmware and EDK2 are well matched when config.txt has the total_mem=2048 assignment.

It may be something that might work for UEFI/FDT mode use as is: The RPi5 firmware probably does live-adjust that FDT information like other models have in the past. But, for FreeBSD, I only expect that UEFI/ACPI would work for booting and operation the RPi5 at this point.

Something to consider. Absent support, it would be good to explicitly document that total_mem should not be used. (Or, maybe that only a few explicitly-listed config.txt assignments are supported and the rest are not, possibly noted via comments in config.txt .)

mariobalanica, 2024-03-16:
EDK2 determines the RAM size based on board revision code. 
I suppose the `memory@0` node in device tree follows `total_mem`, so we could parse that instead.

markmi, 2024-03-20:
FYI:

I used RasPiOS64 (my abbreviation) to set up *.dtd files for 8 GiBytes and for total_mem based 2048 MiBytes via dtc -Ifs -Odtb /proc/device-tree . On FreeBSD I then used dtc to convert to sorted *.dts files and diff'd the files:
```
        memory@0 {
 
                device_type = "memory";
-               reg = <0x0 0x0 0x3fc00000 0x0 0x40000000 0xc0000000 0x1 0x0 0x80000000 0x1 0x80000000 0x80000000>;
+               reg = <0x0 0x0 0x3fc00000 0x0 0x40000000 0x40000000>;
        };
```
Unfortunately, I used a different USB port for the 2 boots, so there was:
```
                        usb {
 
                                lun = <0x0>;
-                               root-hub-port-number = <0x3>;
+                               root-hub-port-number = <0x2>;
                                route-string = <0x0>;
-                               usb-version = <0x3>;
+                               usb-version = <0x2>;
                        };
```
There was one more difference:
```
                nvram@0 {
 
                        #address-cells = <0x1>;
                        #size-cells = <0x1>;
                        compatible = "raspberrypi,bootloader-config", "nvmem-rmem";
                        no-map;
                        phandle = <0x70>;
-                       reg = <0x0 0x3fd165e0 0x38>;
+                       reg = <0x0 0x3fd16600 0x38>;
                        status = "okay";
                };
```