worproject/rpi5-uefi #15: R Pi 5 - Booting Win 11 - No Ethernet

ThePieMonster, 2024-04-14:
I am booting the Pi 5 with Win 11 using the following...

Insert SD card in main PC and select these options:

[image]
[image]
[image]

[image]

After install if I try to boot up the Pi I get a error about os_checker=0 so I copy over the [RPi5_UEFI_Release_v0.3](https://github.com/worproject/rpi5-uefi/releases) which is done by using the [Boot partition mount utility](https://www.worproject.com/downloads#boot-partition-mount-utility).

[image]

I then can boot up into the Windows 11 setup screen however I can not get past the 'connect to the internet' screen as there the ethernet does not appear to be working. I also do not see any activity lights on the ethernet port of the R Pi 5.

FriendlyNGeeks, 2024-04-14:
Google win 11 OOBE disable or get a usb network adapter

onyxprime-serv, 2024-07-12:
This brand worked out of the box for me.
Cable Matters USB to Ethernet Adapter. FBA_202013-BLACK
https://amzn.asia/d/0h1YKLVj