worproject/rpi5-uefi #18: Is there any progress in the development?

Corsair-63, 2024-04-26:
First of all, say thanks for the big effort done in the development of this project.

is it any schedule or improvement in this project?

asking that because should be fine to have a complete support of the video/sound, really particularly important LAN and asking for the moon also Wifi & BT.

Croydon, 2024-04-26:
Yes, thank you very much for the work. Strongly appreciated! ❤️

Adding to the questions above: What is the angle for this effort? Is there a motivation to upstream things eventually? Or is it just experimental, for own purposes, or something like that?

Corsair-63, 2024-04-26:
I own also a RP4 but is not stable at all with W11, purchased some days ago one RP5 and seen that is stable in W11 and having a boat I had ARM64 based computer because of the consumption & performance with w11 and happy with it, but maybe needs replacement and because of the size and afore mentioned matters I have gone to a RP5 that for the moment is giving me very good performance under W11.
but if video & audio plus LAN drivers would be working it should be fantastic, FTB I have installed USB audio & Lan cards and working but would be great to have the LAN and BT working on the own board.

mariobalanica, 2024-04-26:
Hi,

> First of all, say thanks for the big effort done in the development of this project.
> 
> is it any schedule or improvement in this project?
> 
> asking that because should be fine to have a complete support of the video/sound, really particularly important LAN and asking for the moon also Wifi & BT.

@Corsair-63 I assume you're asking about Windows here. I don't really have any plans to further improve support for it. At best I might look at that Ethernet controller when/if I get a chance, but not making any promises.

It all takes a lot of effort and public interest is also fairly low, so I have pretty much no incentive to continue working on this.

I'd strongly encourage you and anyone else interested to kindly contact Raspberry Pi and let them know you care about Windows (and UEFI) support on their boards. If enough people bug them, who knows, maybe one day they will consider investing in it.
Realistically that's the only way to get some real support - community projects can only do so much.

> Yes, thank you very much for the work. Strongly appreciated! ❤️
> 
> Adding to the questions above: What is the angle for this effort? Is there a motivation to upstream things eventually? Or is it just experimental, for own purposes, or something like that?

It started out of curiosity to see what would be possible with this hardware in terms of standardization, so yes, mostly a proof-of-concept. I'm not actively working on it, but this doesn't mean it's dead - contributions are welcome.

About upstreaming:
* The TF-A port is upstreamed, minus one PCI patch that's waiting on review. 

* I've also submitted the patches in my EDK2 base fork back in January, but I haven't heard anything since. The EDK2 review process is slow in general, at times non-existent (understandably, not many maintainers), so I just gave up on that idea.

Corsair-63, 2024-04-27:
> @Corsair-63 I assume you're asking about Windows here. I don't really have any plans to further improve support for it. At best I might look at that Ethernet controller when/if I get a chance, but not making any promises.

Yes, with Windows 11

> I'd strongly encourage you and anyone else interested to kindly contact Raspberry Pi and let them know you care about Windows (and UEFI) support on their boards. If enough people bug them, who [...]

already sent a message to Raspberry. will see if I have any answer or if they consider it.

FrostbyteGR, 2024-05-05:
Let me also thank you a lot; and express my gratitude, for your time and efforts, as well.

> At best I might look at that Ethernet controller when/if I get a chance, but not making any promises.

I'd be very, very, strongly interested, at the very least, in the Ethernet controller too.

My use case is not Windows, but instead a Cloud-like PoE Cluster (w/ MaaS, Juju, K8s, Ceph)

Some information on how it was accomplished on RPi4 could be found here:
[1] https://discourse.maas.io/t/raspberry-pi-4-provisioning-and-kvm-pod-setup/3607
[2] https://discourse.maas.io/t/build-your-own-bare-metal-cloud-using-a-raspberry-pi-cluster-with-maas/5845

Whilst I'm more of a cloud engineer, rather than a developer, I'd like to know if there's any way I can contribute to bring this into fruition (even monetary).

Thanks a lot again!

Trumeet, 2024-05-05:
I'm on Linux, and it would be much better if it supports Ethernet :) Thanks for the hard work though.

ykla, 2024-05-06:
It supports boot FreeBSD 14.0-RELEASE and 15-CURRENT.

primaltechguides, 2024-05-08:
Would we need to wait for GPIO support for an active fan to work properly? As booting UEFI the active fan seems like it is 100% always.

Croydon, 2024-05-17:
```
About upstreaming:

    The TF-A port is upstreamed, minus one PCI patch that's waiting on review.

    I've also submitted the patches in my EDK2 base fork back in January, but I haven't heard anything since. The EDK2 review process is slow in general, at times non-existent (understandably, not many maintainers), so I just gave up on that idea.
```

Could you please give us some links to the pull requests / patch submissions?

heini, 2024-09-08:
> It all takes a lot of effort and public interest is also fairly low, so I have pretty much no incentive to continue working on this.

@mariobalanica, did you think about getting in touch with the [Pi Firmware Task Force](https://github.com/pftf), who already maintain working UEFI implementatios for RPi 3 and 4? Maybe they can take over...