worproject/rpi5-uefi #23: My Raspberry Pi 5 keeps blue screening when booting into Windows 11.

Jpwaters09, 2024-06-01:
# My Raspberry Pi 5 keeps blue screening when booting into Windows 11:
Hi,
I have updated my Raspberry Pi 5 bootloader but it hasn't made any difference.

Stop Code: ACPI_BIOS_ERROR

Kind Regards
Jacob

RASPBPIFOREVER, 2024-06-06:
Please follow the Windows 11 installation thread, especially the following step:
Follow @mariobalanica's instructions for replacing the storahci.sys and stornvme.sys files before moving the NVMe to the Pi: https://github.com/worproject/rpi5-uefi/issues/2#issuecomment-2016505218
This will solve your issue. Regards

Jpwaters09, 2024-06-12:
I have replaced the storahci.sys and stornvme.sys files but it is still giving me the same blue screen.
I have tried to boot from a SD card with Windows and the RPi5 UEFI files on it.

Kind Regards,
Jacob

closed 2024-06-14 by Jpwaters09