worproject/rpi5-uefi #24: rpi5 running win 10- no Ethernet, no WiFi, and no Internet.

escapeZe, 2024-06-15:
No matter what I do I cannot connect to any form of Internet, I've tried re-installing windows, and have also even tried Ethernet to USB 3.0, Please help. ):

Corsair-63, 2024-06-15:
It seems that not yet ready the uefi/drivers for it, in the main page you can see that not yet available.

but you have some option to be connected in windows to the network with the RPi5.

connect your mobile via USB and tell it to do a tethering via USB (sharing connection via USB).
another option is to connect a USB - ethernet adaptor, it will be immediately recognized by windows, and you'll be connected to the LAN (internet)

escapeZe, 2024-06-16:
Thank you, waiting on a USB - ethernet adapter. I will comment back on the results when it arrives.

Govro150, 2024-06-20:
I already tested usb internet. It works

escapeZe, 2024-07-16:
fixed and worked until I added a nvme card using pimo Roni hat off of the PCIe port and it will get to the loading screen and stop at the very end. To be clear I am not booting off of the nvme card, just wanted to use it for storage drive. and when I unplugged the wire to the nvme hat it prepares automatic repair. help?

escapeZe, 2024-07-16:
It works, nvme issues now