worproject/rpi5-uefi #25: Question on the Bootloader

BloodyTakao, 2024-06-26:
Greetings,

I'm planing to install windows 11 on my (soon to have) raspberry pi 5, the sad thing is, I dont have an adaptor or NVME SSD to load the OS on and would use a micro SD card.

Do I need to have the SD card with the bootlaoder all the time in that raspberry pi or can I remove it after I booted it once with the boot loader.
This sadly doenst say in the documentation.

I hope you can answer my question.

Sincerly

Govro150, 2024-06-26:
you can unplug sdcard with UEFI right after logo appeared

BloodyTakao, 2024-06-26:
thank you very much for the fast answer, apreciate it, will do that, once everything arrives <3

Corsair-63, 2024-07-03:
in the first Issue opened is a complete list of how to install win11ARM64 on Rpi5. follow all the instructions and most important the one for the SSD.
[https://github.com/worproject/rpi5-uefi/issues/2]