worproject/rpi5-uefi #26: Won't boot anymore after restart

biast12, 2024-06-30:
Hey i'm running win 11 (arm) on my raspberry pi 5 on an M.2 SSD and it has been working for a few days now with no issues (with restarts in between) tho after i restarted it here some days ago have it been unable to boot at all, it just sets in the boot loader at 100% and does nothing (i had it running all night and it was still stock on 100%) anyone got an idea on what the issue may be or what i could do to fix it?

[image]

biast12, 2024-06-30:
Just to be sure, this is the guide i used to set it up: https://youtu.be/aa0vZHpSX5E?si=qRkORG9yCeQ5rvAA
Tho as said it has been working for a few days now without any issues

huesie, 2024-07-06:
I had an issue specifically booting from NVMe - a Win 11 update replaced the stor*.sys drivers. 

For me it was hung during boot (with the spinning circle from what I can recall) - I fixed it by:

a) mount on Windows PC and updated drivers: https://worproject.com/faq#how-can-i-update-the-drivers

b) replaced stor*.sys drivers for NVMe: https://github.com/worproject/rpi5-uefi/issues/2#issuecomment-2016505218

Hope it helps your situation but please let us know how you get on.

biast12, 2024-07-07:
Thanks for this, i'll test to see if this works, tho one question, i see the git for the drivers doesn't have one for pi 5, should i just use one of the others and if then which one? or can i just skip that part?

biast12, 2024-07-07:
Just tried to see if it would work by just replacing the stor*.sys drivers from https://github.com/worproject/rpi5-uefi/issues/2#issuecomment-2016505218 and it didn't sadly

Also do note that it already worked, just kinda didn't after i restarted the pi

huesie, 2024-07-07:
The Pi3 drivers work for me; and if it broke after running once then could be device incompatibility causing corruption - to diagnose corruption you could try running sfc via dism with the ssd connected to a Windows PC:

https://support.microsoft.com/en-us/topic/use-the-system-file-checker-tool-to-repair-missing-or-corrupted-system-files-79aa86cb-ca52-166a-92a3-966e85d4094e

Start by checking your M.2 carrier board compatibility list and/or try different hardware or you may just have been unlucky on that boot - there's a lot that can go wrong.

Hopefully it's some comfort that many of us run Windows 11 for weeks/months after many reboots, so it does work.  Good luck!

biast12, 2024-07-07:
It did say it had an error when i ran sfc tho when i tried to boot up again did it still not work (also updated the drivers)

is there a way i can get some sort of log file that might show what the issue could be when booting?

i kinda don't wanna start completely over but if there's nothing i can do then i guess i have to

huesie, 2024-07-14:
See https://github.com/worproject/rpi5-uefi/blob/master/README.md#1-prerequisites for advice on getting logs via UART (not tried it myself sorry)

biast12, 2024-10-05:
Ok so i did it all over again some time back and it broke again, so i have given up on running windows on it, i think it might have been one of the things i installed as it seems a bit weird only i had this issue

closed 2024-10-05 by biast12

huesie, 2024-10-05:
Windows update can update drivers @biast12 so it no longer boots - try repeating this step: https://github.com/worproject/rpi5-uefi/issues/2#issuecomment-2016505218

biast12, 2024-10-05:
Already tried that, still didn't wanna boot

huesie, 2024-10-06:
It's interesting to me, as Windows has been astoundingly stable for me, though I'm not running any overclock on CPU or PCIe.

Could you post a bit about your hardware, please @biast12: NVMe brand, model, size, and M.2 adapter? (just interested to know what exact issue you're facing).

I will post some other info, as I've now had success running stock Debian, rEFInd, ventoy, and even Win11 Multi-session just for fun - these Pi5s are a lot of fun!  But sorry to hear you've not had the stability I've seen.

biast12, 2024-10-07:
Raspberry Pi 5 - 8 GB
Micro SD: 16GB Micro SD (only used this for boot)
SSD: 500GB M.2 SSD NVMe - Kioxia EXCERIA
Case: Argon ONE V3 M.2 NVME PCIE Case (the M.2 adapter too)

One thing to note is i could boot first time and after too (not 100% sure if it updated windows before booting the second time)
Some of the programs i know for a fact that i downloaded is Apache, node, Cloudflare tunnel, PHP etc.

I used https://worproject.com/esd to get the ESD Image btw