worproject/rpi5-uefi #30: Win 11 with NVMe PCIe M.2

zipino, 2024-07-18:
Hi everyone, is there a way to install Win 11 on Raspberry Pi 5 with NVMe PCIe M.2 thanks

huesie, 2024-07-18:
Follow instructions in https://github.com/worproject/rpi5-uefi/issues/2#issuecomment-2016505218

zipino, 2024-07-18:
> Segui le istruzioni al [ punto 2 (commento) ](https://github.com/worproject/rpi5-uefi/issues/2#issuecomment-2016505218)

I have to download those drivers right but how do I install Win on NVMe ssd you understand what I mean

huesie, 2024-07-18:
Everything you need is in that issue including links to YouTube videos.

zipino, 2024-07-18:
> Tutto ciò di cui hai bisogno è in quel numero, inclusi i collegamenti ai video di YouTube.

does that tutorial tell you how to install win 11 with NVMe?