worproject/rpi5-uefi #34: Overclocking Windows 11

jusimen, 2024-09-13:
So was able to successfully install Windows 11 on an NVMe SSD following the guide in #2. Now I'm trying to overclock the Pi to use 3GHz but several BSOD appear for different configurations.

Has anyone been able to OC the Raspberry Pi 5 using Windows?

### Here are some of the configurations I've tried and did not work.

```
over_voltage_delta=30000
arm_freq=3000
gpu_freq=1000
```

```
over_voltage_delta=50000
arm_freq=3000
gpu_freq=1000
```

```
over_voltage=3
arm_freq=3000
gpu_freq=1000
```

Marcinoo97, 2024-09-13:
why would u ever want to overclock gpu?

jusimen, 2024-09-13:
This is just what I saw other people were doing while OCing their PIs, don't actually need GPU overclock

Marcinoo97, 2024-09-13:
underclock gpu. it should save power in theory but unsure as its not used by windows anyway.

Marcinoo97, 2024-09-13:
@jusimen were u successful in running rpi5 with this OC but in linux?

jusimen, 2024-09-13:
I didn't even try that as the goal was to use windows. That's because I need to run a program (Rekordbox) that is not supported on Linux. And I've read that the Wine version of the Rekordbox is awful.

I would say I might not even need to OC at all because I now got a fan for the PI and the performance has increased significantly 😅.

Anyways, would be nice to know if there is some OC configuration that works on windows.