worproject/rpi5-uefi #41: A documented future? : Pi500/CM5  and Pi5 firmware requirement?

markmi, 2024-11-26:
Just an FYI: One of the new RPi firmware release notes reports:

> Update recovery.bin to the most recent version required for CM5 and Pi500.
>   (firmware version 2024-10-21)

This suggests that, if rpi5-uefi is to support the Pi500, once it is available, rpi5-uefi may end up needing a update to the RPi firmware it is based on and whatever else goes along with making that work. A `bcm2712-rpi-500.dtb` is published, along with 4 `bcm2712-rpi-CM5*.dtb` files. Likely preliminary content for now?

einsteinagogo, 2025-01-01:
following up from this, does this release 0.3 work with CM5 as is - current ?

mibosshard, 2025-01-03:
> following up from this, does this release 0.3 work with CM5 as is - current ?

at least for me, no. I would also like to know if it should work though...

markmi, 2025-01-03:
There have been no commits after: Mar 16, 2024 . No one with the appropriate background knowledge and interest looks to have worked on it since then. The EDK2 material ended up not going through the process to have it upstreamed into the official EDK2 materials.

For reference . . .

This release (from a microsd card with it):

# mount -noatime /dev/mmcblk0p1 /mnt
# ls -C /mnt/*.dtb
/mnt/bcm2712-rpi-5-b.dtb

Modern (from RaspiOS64 [my abbreviation] media):

# ls -C1 /boot/firmware/bcm2712*
/boot/firmware/bcm2712d0-rpi-5-b.dtb
/boot/firmware/bcm2712-rpi-500.dtb
/boot/firmware/bcm2712-rpi-5-b.dtb
/boot/firmware/bcm2712-rpi-cm5-cm4io.dtb
/boot/firmware/bcm2712-rpi-cm5-cm5io.dtb
/boot/firmware/bcm2712-rpi-cm5l-cm4io.dtb
/boot/firmware/bcm2712-rpi-cm5l-cm5io.dtb

The status relative to the additional *.dtb files and relative to any bcm2712-rpi-5-b.dtb update is unknown/untested so far as I know. No one is working on making sure that any of those would work so far as I can tell, not even if reports of problems were made.

I similar point might apply to updates to the EEPROM content, in part because the loaded *.dtb file content might get live updates before EDK2 is involved.

Nothing claims any of the more modern context is or will ever be supported by someone.

NumberOneGit, 2025-01-10:
The new Pi5 with 2GB RAM also runs on the new chips with d0 stepping (as do all CM5s) and firmware for any Pi related projects needs to be updated for the new chips to work.

omerfsen, 2025-01-25:
I am having same issue. Pi500 black screen but with Pi5 that is ok, though it fails afterwards