worproject/rpi5-uefi #42: Raspberry Pi5 8Gb doesn't load graphical UEFI mode (hdmi blank screen)

kt315ua, 2024-12-18:
HW:
1. Raspberry Pi5 8Gb
2. Kingston NV3 1TB (clean)
3. Kingston Canvas Go! Plus microSDXC 64GB
4. PSU original 27W

Steps:
- prepared SD card with card reader
    - fat16 file system with 32gb size
    - extracted release uefi v0.3 into sd
- turned on with HDMI and UART/USB converter

Actual result:
- UEFI loaded and available only via UART
- HDMI interfaces are blank (but monitor recognize connection)

Also, tested with latest [bcm2712-rpi-5-b.dtb#dc9ec44](https://github.com/raspberrypi/firmware/commit/dc9ec4478ee8aedcfdfc9ddd5a28f1d9029fa153)](https://github.com/raspberrypi/firmware/tree/master/boot) - no result

[image]
[Raspberry_UEFI_boot_no_HDMI.log](https://github.com/user-attachments/files/18188194/Raspberry_UEFI_boot_no_HDMI.log)

kt315ua, 2024-12-18:
Problem found with `framebuffer_depth` parameter.
Release 0.3 contain next parameter value:
```
# Force 32 bpp framebuffer allocation.
framebuffer_depth=32
```

For my Raspberry Pi 5 board changing that parameter to 24 helps: `framebuffer_depth=24`
[image]
[image]

With `framebuffer_depth=16`, picture became pinky 
[image]

PS: I don't know how it will affect Linux/FreeBsd kernel boot processes. But it will fix UEFI boot interface.

StefanFanaru, 2025-01-16:
> Problem found with `framebuffer_depth` parameter. Release 0.3 contain next parameter value:
> 
> ```
> # Force 32 bpp framebuffer allocation.
> framebuffer_depth=32
> ```
> 
> For my Raspberry Pi 5 board changing that parameter to 24 helps: `framebuffer_depth=24` [image] [image]
> 
> With `framebuffer_depth=16`, picture became pinky [image]
> 
> Additionally, there was found comment about issue with this parameter for Pi5: [rsta2/circle#511 (comment)](https://github.com/rsta2/circle/issues/511#issuecomment-2494704597)
> 
> PS: I don't know how it will affect Linux/FreeBsd kernel boot processes. But it will fix UEFI boot interface.

The same for me... setting framebuffer_depth=24 does help (thx) but it looks very sketchy