worproject/rpi5-uefi #43: Raspberry Pi 5B 16 GB doesn't boot.

korhojoa, 2025-01-13:
Trying the UEFI with a 16 GB Pi loads to a black screen,

NumberOneGit, 2025-01-14:
Have you tried to replace the included .dtb file with the d0 firmware file and updated the config.txt file to reference it?  
Once you’ve tried that, you may also want to try changing the frame_buffer value in config.txt to 16 or 24 to see whether that gives you an image.   I haven’t been able to get an OS to load on my CM5 either.   They both use the d0 chips, so are different than the early Pi 5 line-up, but I believe for you that is the main difference, so using the correct firmware file may solve it.

ronaldklop, 2025-01-18:
Had a similar issue on RPI5/16GB. When rpi5-uefi (v0.3) runs the screen turned black.
Setting 'framebuffer_depth=16' in config.txt fixed the display and now I see it loading and starting the code in my FreeBSD EFI partition.
Changing the dtb file (by copying the d0 version over the .dtb in your release) didn't seem to change anything for me about this issue.

blknight88, 2025-01-18:
I can confirm this worked for me as well, "Setting 'framebuffer_depth=16' in config.txt fixed the display"

ronaldklop, 2025-01-18:
BTW depth 24 worked also. Not sure if I spotted any visual difference between the two values.

l00nix, 2025-01-22:
Both 24bit and 16bit framebuffer_depth options work for me to display the boot splash screen. However, system seems to hang and not boot after boot splash timeout or when pressing Enter to boot. System boots fine on 8GB Raspberry Pi 5 with nvme.

Stejs1k, 2025-01-23:
> Both 24bit and 16bit framebuffer_depth options work for me to display the boot splash screen. However, system seems to hang and not boot after boot splash timeout or when pressing Enter to boot. I also added the d0 firmware (bcm2712d0-rpi-5-b.dtb) to the efi partition and added device_tree=bcm2712d0-rpi-5-b.dtb. System boots fine on 8GB Raspberry Pi 5 with nvme.

hey what is d0 firmware, where do I get it?

l00nix, 2025-01-23:
Here is some info on the d0 firmware - https://www.jeffgeerling.com/blog/2024/new-2gb-pi-5-has-33-smaller-die-30-idle-power-savings

This is a pic of the d0 chip version on my 16GB Raspberry Pi 5

[image]

You can get the firmware here https://github.com/raspberrypi/firmware/tree/master/boot - specifically for the d0 Raspberry Pi 5s: https://github.com/raspberrypi/firmware/blob/master/boot/bcm2712d0-rpi-5-b.dtb

Stejs1k, 2025-01-23:
> > hey what is d0 firmware, where do I get it?
> 
> Here is some info on the d0 firmware - https://www.jeffgeerling.com/blog/2024/new-2gb-pi-5-has-33-smaller-die-30-idle-power-savings
> 
> This is a pic of the d0 chip version on my 16GB Raspberry Pi 5
> 
> [image]
> 
> You can get the firmware here https://github.com/raspberrypi/firmware/tree/master/boot - specifically for the d0 Raspberry Pi 5s: https://github.com/raspberrypi/firmware/blob/master/boot/bcm2712d0-rpi-5-b.dtb

I am running on d0 chip with d0 firmware but not my UEFI is not loading, do you know what to do?

l00nix, 2025-01-23:
> I am running on d0 chip with d0 firmware, but now my UEFI is not loading at all, do you know what to do?
edit: I renamed d0 firmware as normal firmware, but windows is still not loading, freezing on boot.

This is why we are having this conversation ;-) looks like the d0 version of the Pi 5 boards is not supported currently by the UEFI firmware. I am in the same boat. Maybe someone will find a work around or update the code.

blknight88, 2025-01-23:
I possibly have the same issue as I can't get the latest VMware ESXi ARM Fling ISO to boot either on my 16 GB Pi 5