worproject/rpi5-uefi #47: Pi5 does not boot past grub/uefi (EFI stub: exiting boot services...)

Cajunvoodoo, 2025-01-22:
After fixing the framebuffer issue, I was able to see the UEFI screen and shell. However, when I attempted to boot anything (nixos, ubuntu), I would get the following behavior:
- if no grub (i.e., nixos), hitting "enter" on the boot menu resulted in EFI errors
- if grub, it will show up before sitting on a solid cursor.

The EFI issue is the following:
```
EFI stub: Booting Linux Kernel...
EFI stbu: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Generating empty DTB
EFI stub: Exiting boot services...
_
```

Currently, I am cross-compiling the linux kernel with ACPI debugging on, but until that is done, I am not sure where to go from here. Any help is appreciated.

huesie, 2025-01-27:
Hi @samip5 - I'm collecting notes on repro steps - on 16GB (D0 stepping) running DEBUG build and crash is the same for me even with updated blob.

I can't reproduce this as updated DTB didn't seem to change crash - can you share some additional info on your hardware and blob download url please?

samip5, 2025-01-27:
> crash is the same for me even with updated blob.

Do you happen to have the debug probe? https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html
If you do, it's advised to look at uart.

I think the issue here is that the kernel one is trying to boot doesn't support Pi5.

@Cajunvoodoo Which NixOS version was tried and how did you create the boot media?