worproject/rpi5-uefi #51: Address space d0

NumberOneGit, 2025-02-03:
For those who are more versed in bare metal than I am, we are using:
device_tree_address=0x1f0000
device_tree_end=0x210000

On the new d0 chips, do we know that the address location or the length of the new dtb are not causing us issues?  For example, if the dtb is longer than the one used for the earlier Pi 5’s or if the address start space is already occupied by something else related to the boot process?

NumberOneGit, 2025-02-04:
Using the correct dtb along with 24 for the framebuffer I can install ESXi.  This was my first attempt to do that (I tried Windows before).  Unfortunately, the framebuffer does persist through the install.  I'm hoping once it's fully installed and updated something will change, but I won't be able to try that until I have a USB to ethernet adapter (ESXi doesn't work with Pi's port).  I updated to a newly released eeprom, which may be what led to loading past UEFI.

rotechhype, 2025-02-04:
Hello, 

I am trying to do the same, install ESXi on Rpi5 with d0 chips. Can you provide more details about the steps you took in order to make it boot?

Thanks

rotechhype, 2025-02-04:
I have tried with d0 dtb file and still didnt manage to install it. 
On ESXI Arm 1.15 i get following error : 
<6>Loading /imgdb.tgz
<6>Loading /imgpayld.tgz
<6>UEFI Secure Boot is not enabled
<6>Located crypto module: VMware's ESXboot Cryptographic Module, v1.0
<6>Shutting down firmware services...
PANIC at PC : 0x0000000000006414

If i try to install ESXi ARM 2.1, I get this error:
<6>Loading /imgpayld.tgz
<6>Located crypto module: VMware's ESXboot Cryptographic Module, v1.0
<6>Shutting down firmware services...

SError Exception at 0x00000000383C540C

SError Exception at 0x00000000383C540C

Did you manage to get past this and installl ESXi? If so, what version did you use?

NumberOneGit, 2025-02-04:
I haven’t fully installed yet.  I got stuck at the stage where ethernet is needed to complete the install.  My USB/ethernet adapter arrives tomorrow so I will let you know then.  I did receive an SError as well, but assumed that was related to my stalled install.  I will update when I can try again.   The update was mainly to say that I had moved beyond UEFI to the loading process.

NumberOneGit, 2026-10-29:
I posted this in the first week discovering this project.   The issue was with D0 boot was with pinctrl.

closed 2026-10-29 by NumberOneGit